rbaron/b-parasite #149: Low battery voltage

andresp, 2023-07-10:
Are levels for medium to low/critical battery voltages known? At what voltage is a battery generally considered depleted?

oleo65, 2023-07-15:
In my experience this depends highly on the devices you run with the battery.

For the parasites mine usually stop working when the battery voltage is somewhat around 2.6 V. The coin cell is not able to provide the current spikes any more needed for transmitting the data and the device reboots / power cycles because of that.

I believe the MCU can go even lower for normal operation, but again this also depends on how it is used and what devices are attached to it.

jowi24, 2024-03-23:
Do you recognize a drift in moisture measurements when the battery voltages gets lower?

rbaron, 2024-03-25:
@jowi24 we try to estimate a bit of the effect of the battery voltage on the moisture measurements. In [this comment](https://github.com/rbaron/b-parasite/issues/79#issuecomment-1650278703) and linked spreadsheet there's a very simple model of ADC readings vs. battery voltage.

The coefficients of that model is what we use to interpolate the moisture values during operation, given the current battery state. It's a pretty simple, linear-ish, single variable model.

jowi24, 2024-03-29:
Great, I have a look into the referenced documents. Thanks! 👍