rbaron/b-parasite #156: Additional component files for 2 side printing and the E73-2G4M08S1CX

ryanrolds, 2023-08-14:
I recently received a batch of 5 v2.0.0 devices from JLCPCB. This batch required no soldering. It used 2 sided placement and the E73-2G4M08S1CX. I've been using the finished devices for a couple weeks now and things look good. I intend to order a batch of 50 to build confidence in the changes.

I would like to contribue these changes upstream. Do you want them? Should I create a new placement file/CSV and add a blurb to the docs? 

Thank you for the project, it's affored my first foray into making embedded devices.

rbaron, 2023-08-14:
Thanks @ryanrolds, it always makes me happy to hear these first timer success stories. Consider also popping and saying hi in #19.

As for your changes, I try to be very diligent with things in the main branch when it comes to testing/calibration/power profiling. If your changes are only about adding a bottom placement file (which should be for the battery clip only) and you still have the exact same file you used when placing your order, I'm happy to merge it.

wrobelda, 2023-10-13:
@ryanrolds have you had a chance to look into it? I am looking to place an order with JLCPCB, too, and they indeed suggested a lot of replacements, including E73-2G4M08S1CX. 2 sided placement would be a nice bonus as well.

ryanrolds, 2023-10-13:
https://github.com/ryanrolds/b-parasite/commit/104610e27298a6ae5f9e0b99a2ad1fbbb8e3a2db. I don't guarantee the changes are correct, double-check things. I want to do one more batch to double-check those values before I start encouraging their use. The 5 devices work fine but require an antenna. I've used [these](https://www.amazon.com/gp/product/B09BT6VC92) successfully.

wrobelda, 2023-10-13:
> The five devices I received work fine but require an antenna. I've used [these](https://www.amazon.com/gp/product/B09BT6VC92) successfully. 

I saw people modifying the original NORDIC chip (the one without X suffix) to add an external antenna, anyway, so one could consider that an advantage actually

ryanrolds, 2023-10-13:
Let me know if you find a commercially available antenna that isn't as large.

victorhooi, 2023-11-26:
I just found this project recently, and I'm looking to put my first order with JLPCB for it as well =).

I had a quick question - I did see on the eByte website, that there's two similar-sounding chips:

* [https://www.cdebyte.com/products/E73-2G4M08S1C](E73-2G4M08S1C) - This is what's included on the current BOM for this project
* [https://www.cdebyte.com/products/E73-2G4M08S1CX](E73-2G4M08S1CX)

Does anybody know what the difference is between the two chips? (Other than the addition of an external antenna port on the E73-2G4M08S1CX?)

I checked on https://jlcpcb.com/parts/componentSearch - and it seems both might actually be in stock right now:

[image]

Should I go ahead and order one with the E73-2G4M08S1C now, or is there some advantage to swapping out for E73-2G4M08S1CX?

rbaron, 2023-11-27:
@victorhooi thanks for the suggestion, I added it to the wiki.

I haven't built any versions with an external antenna, but you can check some finished photos in our nice Community Gallery #19:
- [Example 1](https://github.com/rbaron/b-parasite/issues/19#issuecomment-1223632233)
- [Example 2](https://github.com/rbaron/b-parasite/issues/19#issuecomment-1812951503) from @ryanrolds above

victorhooi, 2023-11-28:
@ryanrolds Do you happen to have the STL files for the case you showed in [example 2](https://github.com/rbaron/b-parasite/issues/19#issuecomment-1812951503). It looks pretty neat =).

Are you worried at all about water spray etc, with this design? And how does it work in terms of the light sensor - doesn't the case block that from receiving light?

Also - did you see a noticeable increase in range, using these external antenna?

ryanrolds, 2023-11-30:
I do, but I'm not happy enough with them to share them. The outer "jacket" doesn't allow light or airflow. I will follow up when I'm further along.

ryanrolds, 2023-11-30:
I believe there is a modest increase in range with the external antenna, but I have no evidence to back it up. I've found that some kind of case is needed to prevent watering the plant from causing problems with the device.

ryanrolds, 2023-12-14:
I'm happy to contribute the "harness" I made, not the outer case. I use the harness as a platform and create external cases that clip against it.

[harness_v0_1_0.zip](https://github.com/rbaron/b-parasite/files/13678497/harness_v0_1_0.zip)