rbaron/b-parasite #158: Device not found when scanning BLE

dvandereep62, 2023-09-18:
I have successfully built one device which is working great. I have just built a second device which seems ok and the code flashes fine but when scanning BLE with ESPHome I can't see it.
No idea what the issue could be.

rbaron, 2023-09-18:
@dvandereep62 have you taken a look at the logs of the faulty board?

dvandereep62, 2023-09-18:
> @dvandereep62 have you taken a look at the logs of the faulty board?

Not 100% sure what I'm doing bit using the RTT Viewer I have seen this.

[00:00:00.002,685] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.002,716] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.002,746] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 224.11902 Build 2231721665
[00:00:00.003,540] <inf> bt_hci_core: Identity: C4:88:A6:82:2A:AD (random)
[00:00:00.003,570] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x1077, manufacturer 0x0059
[00:00:00.003,601] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0x1077
[00:00:00.003,601] <inf> ble: MAC Address: c4:88:a6:82:2a:ad
[00:00:00.844,421] <err> i2c_nrfx_twi: Error 0x0BAE0001 occurred for message 0
[00:00:00.844,451] <err> shtc3: Error -5:  in C:/Nordic/code/nrf-connect/prstlib/src/shtc3.c:19
[00:00:00.844,451] <err> shtc3: Error -5:  in C:/Nordic/code/nrf-connect/prstlib/src/shtc3.c:27
[00:00:00.844,482] <err> sensors: Error -5:  in C:/Nordic/code/nrf-connect/prstlib/src/sensors.c:15
[00:00:00.844,482] <err> main: Error -5:  in ../src/main.c:27

rbaron, 2023-09-18:
It looks like an issue with the temp/humidity sensor, similar to https://github.com/rbaron/b-parasite/pull/150#issuecomment-1683548488. I unfortunately haven't been able to reproduce it.

Was your faulty board also hand assembled? Since one of your boards work, my first guess it's hardware related, but it might be some SHTC3 quirk that shows up infrequently. You can try implementing the same workaround as the linked comment as a proof of concept, or try to touch up the solder joints to the module <-> SHTC3 IC. Please let us know how it goes.

dvandereep62, 2023-09-19:
Thanks for the help. I tried to reflow the solder again but made no difference. I also tried the wake retry as you suggested and that also didn't help. I ended up removing the sensor completely from the code and it now works. I'll remove the chip as it seems to have failed or is faulty and replace with another see how that goes. Thanks for the assistance.

closed 2023-09-19 by rbaron