rbaron/b-parasite #159: Multible Sensors with on MCU

TheSmartGerman, 2023-10-13:
@rbaron 
will it be possilbe to controll serveral Sensors with one MCU?

Just a thought:
use one MCU, Light, Temp sonsor and multiplie the moisture sensor -> make a wired connection max ~ 50cm. 

This will make it able to monitore several plants at one spots.

At one spots only the moisture sensore have to be multiplied.  Is there any HW or SW limitations to implement e.g 3 or 4 "sensors"

Thx

rbaron, 2023-10-14:
It's definitely be possible to multiplex a single MCU to multiple capacitive sensors. It would require some nontrivial software and hardware redesign though.

TheSmartGerman, 2023-10-17:
thx, I'll have a deeper look in your work. the HW shouldn't be a problem for me. For the purpose (plant monitoring) it's also not neccessary to monitor multible sensor at the same time. I guess, to work with ESPHome the protocoll have to be extended to suppor more then one moisture reading? Or, if you "fake serveral" sensor, one mcu need serveral mac adress?

rbaron, 2023-10-17:
I took a quick look at the [BTHome v2 docs](https://bthome.io/format/), specifically the `Multiple measurements of the same type` section:

> If you want to send multiple measurements of the same type, e.g. three temperatures, you can just add multiple measurements of the same type to the payload. A postfix will be added to the measurement name (e.g. temperature_2) in the order of which you define the measurements. Note that this implies that you will need to use the same order in each advertisement, to prevent measurements being assigned to the wrong entity. If only one measurement of a certain type is sent, no postfix will be used.

I believe you could just attach multiple moistures values in the same advertisement packet. At 4 bytes per reading, you could fit 5 sensors there without any hacks.

TheSmartGerman, 2023-10-19:
> This would be particularly interesting if one could couple it with these super cheap Aliexpress sensors readily available:
> https://www.aliexpress.com/item/32832538686.html

In fact i did some research the last few days and ordered this sensor's. 
With the TLC555 you can run directly from 3V, if you like you also can modify too do frequently counting instead of a analogue voltage...