rbaron/b-parasite #161: Idea: a 'light' version without the sensor itself?

wrobelda, 2023-10-19:
There seems to be a highly-rated soil moisture sensor that comes readily available on Aliexpress for as little as 0.40 USD:
https://www.aliexpress.com/item/32832538686 

Would it make sense to design a version of b-parasite that would only come with the circuitry required to connect to the above sensor's interface? I imagine most people use a 3D printed box, so putting these two together would significantly:

1. lower the cost
2. improve repairability, should the sensor eventually corrode or otherwise failed 
3. potentially enable other version of sensors to be used, like longer/deeper ones

I also wonder if would be worth contacting the seller/manufacturer to see if they wouldn't be interested in selling readily-made b-parasite? For what it's worth, I found another product that is pretty much a complete product, except with offline sound notifications:
https://www.aliexpress.com/item/4000586280615.html

alexhk, 2024-01-06:
Your link doesn't work (.html missing). Here is the link:
https://www.aliexpress.com/item/32832538686.html

I tend to agree, a separate RC circuit  board sounds like a good idea. I am new to this kind of sensor, but if they are readily available the existing boards could be used for b-parasite or @rbaron could simply separate his RC circuit into a second PCB which could be ordered (in higher quantities) together with the main PCBs. 

I also would like to see longer/deeper sensors (or simply submerge the "little" ones deep into the substrate if they could be connected with a wire).

jrhbcn, 2024-01-07:
I think it is an excellent idea. We could design a 3D printed case that joins the two "parts" similarly to the actual b-parasite or use it separately with longer cables as @alexhk suggests. My +1.

SchmidL, 2024-03-11:
I discussed it with @rbaron already few years ago in #11 

I planned to do it soonish but life had other plans! Love to see the idea to wear some fruits

gabmert, 2026-09-19:
Someone did already do something similar, probably you can take inspirations from it https://github.com/smartboxchannel/EFEKTA_E73C_PWS_MODULE

If you buy the cheap sensor board, here's a nice heads-up video of possible problems by Martin McMaker / Flaura https://www.youtube.com/watch?v=IGP38bz-K48