rbaron/b-parasite #171: Identify BLE device

andresp, 2023-12-11:
I've been flashing a bunch of b-parasites in bulk and was wondering if there is a quick way to read out the BLE MAC?

As a feature request I would suggest implementing an "identify" command that makes the led flash, if that's possible through BT-Home. This could be helpful when provisioning and locating sensors.

rbaron, 2023-12-12:
Hey @andresp,

In the ble sample, the MAC address is printed via RTT logging on first boot. This is what I use to check the MAC address. You can also choose your own MAC address via compilation flags. 

The identify service is a good idea. The only issue is that currently, to squeeze the battery life to the max, we only advertise  for a bit and shut the radio off completely. For an identify service, we'd need to advertise more often and also advertise more non-sensor data, which also consumes extra battery. As most people additionally use a case,  the LED is not usually visible for the most part of its operation. I would be fine leaving this behind a config flag, off by default, or having a timer for 10-30 minutes in which the service is active. If someone wants to take a stab at it, I'm happy to review it.

Nerdboy2, 2024-03-07:
I do it with the nRF Connect phone app. 
I just start a scan in the phone while the sensor is flashing and when it starts to operate the phone/app will pick it up and show the MAC. 
Super simple. Altough the RTT works aswell

fabian727, 2024-05-29:
there is a button on the device. I didn't go deep into the code. Couldn't it be used to wake up the board and send the MAC?