rbaron/b-parasite #175: Bogus illuminance and moisture readings

drspangle, 2023-12-28:
Hey all, been a while. I recently came back to working on this project and decided to bring up one of my v1.2.0 b-parasites. Everything looks good, I've flashed it with the `ble` firmware sample using `bparasite_nrf52840@1.2.0` and it was detected immediately by Home Assistant. Great! I am getting good humidity and temperature readings, and the voltage readings seem fine as well. All three values look nominal when powered by coin cell or the debugger using pogo pins. 

However, the illuminance values are stuck at 0 lx and the moisture values are stuck at 100%. It looks like all the components on the board are in good shape and I'm testing on my bench, so there's no grime or corrosion to speak of which would affect this. I expected the moisture readings to be 0% and the illuminance values do not vary regardless of lighting conditions.

I'm at a bit of a loss here. Maybe there's a calibration routine I've missed on, and maybe some simple probing on the board will give me some hints as to what's wrong? Any suggestions?

drspangle, 2023-12-28:
I've managed to get my debugger working and am reading output from the `soil_read_loop` firmware sample, here's what it looks like for a few seconds' worth of output from having the sensor suspended in open air, powered by the debugger's 3v3:

```
[00:03:54.424,407] <inf> main: 3.29;0
[00:03:54.924,560] <inf> main: 3.30;0
[00:03:55.424,713] <inf> main: 3.30;4294967295
[00:03:55.924,835] <inf> main: 3.30;4294967295
[00:03:56.424,987] <inf> main: 3.30;4294967295
[00:03:56.925,140] <inf> main: 3.30;0
[00:03:57.425,262] <inf> main: 3.29;4294967295
[00:03:57.925,415] <inf> main: 3.30;4294967295
[00:03:58.425,567] <inf> main: 3.30;0
[00:03:58.925,689] <inf> main: 3.29;4294967295
[00:03:59.425,842] <inf> main: 3.30;4294967295
[00:03:59.925,994] <inf> main: 3.30;4294967294
[00:04:00.426,116] <inf> main: 3.30;0
[00:04:00.926,269] <inf> main: 3.31;4294967294
[00:04:01.426,422] <inf> main: 3.30;0
[00:04:01.926,544] <inf> main: 3.30;4294967295
[00:04:02.426,696] <inf> main: 3.30;0
[00:04:02.926,849] <inf> main: 3.30;0
[00:04:03.426,971] <inf> main: 3.30;0
[00:04:03.927,124] <inf> main: 3.31;4294967295
[00:04:04.427,276] <inf> main: 3.30;0
[00:04:04.927,398] <inf> main: 3.23;4294967295
[00:04:05.427,551] <inf> main: 3.25;4294967294
[00:04:05.927,703] <inf> main: 3.26;1
[00:04:06.427,825] <inf> main: 3.22;0
[00:04:06.927,978] <inf> main: 3.23;1
[00:04:07.428,131] <inf> main: 3.25;4294967294
[00:04:07.928,253] <inf> main: 3.23;4294967295
```

This seems pretty bogus to me. The "battery" readings from the 3v3 seem nominal, but the soil probe values seem to be totally garbage. I'm not sure what to do with this. Does this mean I've got a bad board (since the sensor is ostensibly the board itself)?

rbaron, 2023-12-28:
Hey @drspangle,

I would try, sorted by likelihood, difficulty:
1. Double check the module you have and the module you compiled for. The 1.2.0 hardware has a fast discharge circuit whose pins are different between modules. That could explain the error, since the capacitive pads never discharge
2. Look for shorts/cold solder joints, specially between adjacent module pads. I would pay special attention to pins 2 & 3, as they are related to the moisture sensor in the 1.2.0 version (`FAST_DISCH_EN` and `SENS_OUT` -- these are the second and third physical pads on the `nRF52840` module):

[image]

[image]

A few follow up questions:
1. Did you assembly this board yourself?
2. Do other boards from the same batch work with the same code?
3. What module are you using?
4. Any coating?

Please feel free to upload a photo of the board. Maybe we can spot something wrong with the components.

drspangle, 2023-12-28:
Thanks for the reply @rbaron. 

>     Did you assembly this board yourself?
>     What module are you using?

No, I used JLCPCB and installed only the processor module. Here is a picture of the module, I can't seem to identify what the actual chipset is and I am hoping you might be able to identify it before I can.

[image]

>     Do other boards from the same batch work with the same code?

I have not attempted to bring up any other boards from this particular batch, so I can't use this to rule out anything sadly.

>     Any coating?

Not yet. I wanted to test a couple from this batch before I began coating them using your tried and tested method.

drspangle, 2023-12-28:
@rbaron I think I may have fixed a cold solder joint on pin 3 as it now has good continuity with R6. I also seem to have good continuity between pin 2 and R3. I also checked pin 15 and it has good connectivity with R4. That said, I'm still getting confusing readings from soil_read_loop (though they aren't zero constantly):
```
*** Booting Zephyr OS build zephyr-v3.5.0-3496-g87d056bd31b1 ***
[00:00:00.347,015] <dbg> pwm_nrfx: pwm_nrfx_set_cycles: channel 0, pulse 16, period 32, prescaler: 0.
[00:00:00.347,015] <inf> main: input_voltage;soil_adc_output
[00:00:00.847,167] <inf> main: 3.30;2
[00:00:01.347,320] <inf> main: 3.30;2
[00:00:01.847,442] <inf> main: 3.31;2
[00:00:02.347,595] <inf> main: 3.31;1
[00:00:02.847,747] <inf> main: 3.30;2
[00:00:03.347,869] <inf> main: 3.30;3
[00:00:03.848,022] <inf> main: 3.30;1
[00:00:04.348,175] <inf> main: 3.31;1
```

When I run the modified version as above, `soil_val_mv / 1000.0` is still always zero. There doesn't seem to be any variance when I squeeze the bottom of the PCB where the sensor is (in either the original soil_read_loop or the modified version). Am I misinterpreting something or missing something obvious?

rbaron, 2023-12-28:
Nice! I'm glad to hear you got it working. I will close the issue.

For coating and calibration, I've only ever used acrylic spray coating, and from my manual tests the coating layer is so thin that I saw little change in the capacitance measurement with and without coating, so I didn't bother maintaining two sets of calibration coefficients.

If you want to explore this further, I would compare the `soil_read_loop` readings from an uncoated board and a coated board with your preferred method. If they differ too much, you'll ideally have to calibrate it. I wrote [this](https://github.com/rbaron/b-parasite/issues/79#issuecomment-1637418626) and [this](https://github.com/rbaron/b-parasite/issues/79#issuecomment-1650278703) comments outlining how I did it.

It's up to personal preference, but depending on how much they differ (e.g.: up to 10-20%), I would not bother recalibrating it. Some other effects, like potential drift (#2) and positioning (#104) probably cause more variance in readings than fine tuning by calibration can resolve in my experience.

closed 2023-12-28 by rbaron