rbaron/b-parasite #181: Repeating measurement

Nerdboy2, 2024-02-16:
Hello!
I finally got my first stick up and running but it behaves a little weird I think. 
The last few days it seems to repeat the same values over and over again. 
This is the result in HA 
[image]

The voltage is stable at ~3.1V, I have not watered it, I have a polyurethane coating on it (Boat varnish) and I use the bparasite encoding and an ESP32 as a bridge. 

Hope I got all the necessary information... Any clues what could cause this and what remedy there is?

Nerdboy2, 2024-03-06:
So far I only have this board up and running.
It's the 1.2.0 version running the BLE sample with standard settings. 
I run the latest version of ESPHome and HA.

All values from the sensor is the same. However, I have a DHT sensor on the ESP32 running the bridge and that works fine, no issues what so ever. 

The issue is resolved temporarily when I press the restart button on the ESP32. ie, the bridge starts sending the correct values again.  

My first thought was a RAM-issue but then the DHT sensor should show the same behavior. 
So my current theory is some kind of problem with the ble_tracker in ESPHome.

So far I have found no pattern to this issue. It seems to happen at random.

Nerdboy2, 2024-03-06:
I left the log open for a while, then this just happened. 

`[20:53:58][E][esp32_ble_tracker:291]: ESP-IDF BLE scan never terminated, rebooting to restore BLE stack...
[20:53:58][I][app:127]: Forcing a reboot...
[20:53:58]Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
[20:53:58]
[20:53:58]Core  0 register dump:
[20:53:58]PC      : 0x4019be73  PS      : 0x00060735  A0      : 0x80119155  A1      : 0x3ffb78b0  
WARNING Decoded 0x4019be73: esp_bt_controller_shutdown
[20:53:58]A2      : 0x3ffc7394  A3      : 0x00000000  A4      : 0x00000003  A5      : 0x00060923  
[20:53:58]A6      : 0x00060920  A7      : 0x00000001  A8      : 0x3ff71000  A9      : 0x38000000  
[20:53:58]A10     : 0xffc00000  A11     : 0x3ff51110  A12     : 0x00060920  A13     : 0x00060d23  
[20:53:58]A14     : 0x007b7568  A15     : 0x003fffff  SAR     : 0x00000003  EXCCAUSE: 0x00000005  
[20:53:58]EXCVADDR: 0x00000000  LBEG    : 0x40090e4c  LEND    : 0x40090e57  LCOUNT  : 0xffffffff  
[20:53:58]
[20:53:58]
[20:53:58]Backtrace:0x4019be70:0x3ffb78b00x40119152:0x3ffb78d0 0x4008ca5c:0x3ffb78f0 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4019be70: esp_bt_controller_shutdown
WARNING Decoded 0x40119152: bt_controller_shutdown at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/controller/esp32/bt.c:1806
WARNING Decoded 0x4008ca5c: ipc_task at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_ipc/src/esp_ipc.c:77
[20:53:58]
[20:53:58]
[20:53:58]Core  1 register dump:
[20:53:58]PC      : 0x40202bf2  PS      : 0x00060f35  A0      : 0x800f6e18  A1      : 0x3ffbd790  
WARNING Decoded 0x40202bf2: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[20:53:58]A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060023  A5      : 0x80000000  
[20:53:58]A6      : 0x007bf0b8  A7      : 0x003fffff  A8      : 0x00000000  A9      : 0x00000004  
[20:53:58]A10     : 0x3ffbd2ec  A11     : 0x80000001  A12     : 0x80096a30  A13     : 0x3ffcd6a0  
[20:53:58]A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
[20:53:58]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
[20:53:58]
[20:53:58]
[20:53:58]Backtrace:0x40202bef:0x3ffbd7900x400f6e15:0x3ffbd7b0 0x40094b94:0x3ffbd7d0 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40202bef: cpu_ll_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:837
WARNING Decoded 0x400f6e15: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/freertos_hooks.c:63
WARNING Decoded 0x40094b94: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3973
[20:53:58]
[20:53:58]
[20:53:58]
[20:53:58]
[20:53:58]ELF file SHA256: 0000000000000000
[20:53:58]
[20:53:58]Rebooting...
[20:53:58]ets Jul 29 2019 12:21:46
[20:53:58]
[20:53:58]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:53:58]configsip: 0, SPIWP:0xee
[20:53:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:53:58]mode:DIO, clock div:2`

Any clue what to do with it? Caus I have no idea

rbaron, 2024-03-07:
From the ESPHome logs you posted, it looks like the ESP32 is having issues with the ble scanner. There's a watchdog timer ([ESPHome code here](https://github.com/esphome/esphome/blob/13736b5c5777d37470da8696163a3a7dd69c421b/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp#L290) that reboots the whole ESP32 if the BLE scanner doesn't successfully restart. It's difficult to pin point a reason from a distance, but it could be hardware (e.g.: an old ESP32 board, or issues related to different variants) or software (scanner settings, too many BLE devices being scanned).

I would try to simplify the ESPHome setup to the bare minimum, with just the `bluetooth_proxy` and remove every other feature/sensor. I'd pay extra attention to keep the default BLE scanning window and interval, as running BLE and WiFi at the same time can get finicky.

As a second step I would get a different ESP32 board and re-run the simplified test setup. I myself am running a few ESP32-S3 boards around the house.

Nerdboy2, 2024-03-07:
Alright thanks, Ill try that! 

I am aware this is not really related to the bparasite, but is it possible to have the ESP32 deliver the data through USB? My initial googling says no but I only found old threads on the HA forum.

rbaron, 2024-03-07:
While testing I usually set the log level to `DEBUG` on the component I'm interested in. Then I see the messages via the USB-UART adapter. But I suspect that it won't help a lot in this case, since the issue seems to be in the BLE scanner. I think you'd see something similar to the log you posted above: ESP32 rebooting due to the watchdog triggering.