rbaron/b-parasite #183: Battery voltage drop after update

ericu65, 2024-03-03:
Hi,

I've been using 5 boards (fully JLCPcb  assembled except the backside) of version 2.0.0 for the last 7 months with almost no issues using BTHomev2. Recently I've noticed two of my boards have started corroding so I cleaned them up, removed the corrosion, recoated them in two layers of Plastik 70 and replaced with a new battery. I left them in water for a week to confirm default calibration was still usable but it only went up to 90% moisture reading so I decided to update them to the latest firmware. I cloned the latest master, updated the toolchain from 2.4.0 to 2.5.2 and built them with the following config without connecting VCC and using the on-board battery:
[image]

While for the prior week the voltage had been pretty stable following the battery change:
[image]
Immediately after flashing, both boards' battery measurement dropped significantly within a couple of 10m update cycles:
[image]

I measured the same voltage with a multimeter (although about 50mV higher as I expect the voltage is measured by the firmware under load).

Strangely after a week first one of the boards, then the other. started increasing its voltage magically (again, confirmed with a multimeter). They are both still just in a cup of water so no change to the environment:
[image]

Any idea what could explain this behaviour? Initially, I thought the battery consumption during flashing was the culprit but the voltage dropped within 30 minutes after disconnecting and the latest increase is not explainable by this. 

PS. Wonderful OSHW you've created here. Thank you.

rbaron, 2024-03-04:
Hey @ericu65,

This is both interesting and intriguing, and I can't say I've experienced a similar issue yet. Specially seeing the battery voltage going up! Given you confirmed the odd behavior with a multimeter (in-circuit I'm assuming), one wild idea is that it could be related to ambient temperature. I'm not super convinced, because the temp variation would have to be kinda large.

For example, check out this chart from [a CR2032 datasheet](https://www.farnell.com/datasheets/1496885.pdf):

[image]

There is a correlation between temp and voltage reading, which could explain the first graph if in your test you used colder tap water, for example -- but I would expect the voltage to slowly go back up as the temp settles back. Or it could explain the second chart (voltage going up) if you moved the sensor indoor from outside.

Do you recall doing any of these?

A couple of other (relatively far fetched) ideas:
- Temporary partial short caused by water around the battery clip or some exposed pads
- There is a known "initial drift" [see wiki](https://github.com/rbaron/b-parasite/wiki/What-to-Expect-from-the-Sensor#initial-drift) in the moisture reading in new sensors, though I haven't seen it affecting the battery reading

ericu65, 2024-03-05:
Hi @rbaron 

Thanks for looking into it. I think this rules out a known change in the way the voltage ADC pin is read. That was my main concern, change from version to version, which seems to not be the case. Out of the 5 these two have given me the most trouble regarding excessive battery usage often needing to change them every 2 months so it's likely something electrical that would explain these swings I see here too. Seems to be pretty stable now and I've put them in soil.
[image]

To your good points:
- temperature change would surely explain this but not the case here, have been inside in the same place the whole time and the room's temperature changes do not correlate to the voltage changes (see above - obsessive data collection is real), and the water wasn't replaced as far as I can remember - and the SHTC3 doesn't show any big swings either;
- voltage was measured in-circuit but not under TX load;
- I've also seen the drift for moisture reading but only in soil, not in water - though not for voltage until now;
- an intermittent low resistance short either at the pins or at the battery clip is most likely the real reason as per the previous experience with these two boards.

Notice the calibration referencing VCC working beautifully with no change to moisture during the voltage swings - really nicely done: 
[image]

Since there's nothing to be fixed I will close this shortly. 

Thank you