rbaron/b-parasite #190: Building zigbee fails with FATAL ERROR

nl0pvm, 2024-04-21:
The ble build works great have not worked out how to flash but anyway. I hoped to build and flash the zigbee build but no luck so far.:

[build log.txt](https://github.com/rbaron/b-parasite/files/15053060/build.log.txt)

oleo65, 2024-04-22:
In my opinion there is some changes in the underlying build architecture of the nrf toolkit and some dependencies are upgraded.

Please target v2.5.2 of the nrf toolkit for now, which works for zigbee compilation.

There might be an easy fix for this, but this might be something to pick up for someone with more C skills. 😉

nl0pvm, 2024-04-23:
Thanks will try that and report back here

greenshado, 2024-04-26:
Had the same problem, downgrading to v2.5.2 solved it.