rbaron/b-parasite #199: Add a ducumentation on how to calibrate the sensor

Exeu, 2024-05-09:
Hi,

i don't know, but i think it would be beneficial to have a small guide how calibration can be done.
I think there are more or less some default configs in the BLE-FW but currently it is not clear to me how to get the values for instance for the coeffs.

[image]

First of all where i can configure those Coeffs?
I found one place in an overlay-file.

[image]

The values i got are times 1000 yes, but how you got those values? There are some negative values aswell.
I am a bit lost in trying to interpret them somehow ^^

Furthermore i would like to determine my own coeffs - Because i think those will depend on each board setup slightly.

Thank you

rbaron, 2024-05-09:
Hi @Exeu,

Indeed we need better docs for calibration. Or even better, some automation for it. Meanwhile, I wrote a quick and dirty expanation of what I did in #195, https://github.com/rbaron/b-parasite/issues/79#issuecomment-1637418626 and https://github.com/rbaron/b-parasite/issues/79#issuecomment-1650278703.

SchmidL, 2024-08-01:
maybe https://github.com/SchmidL/b-parasite_autocali is interesting as explained in https://github.com/rbaron/b-parasite/issues/79#issuecomment-2262838017