rbaron/b-parasite #200: Suggestion: Buy a ton of boards in bulk and sell them to us

jordanlambrecht, 2024-05-10:
I (and a ton of other people most likely) don't want to deal with the whole manufacturing process of getting a custom board made. Totally fine doing the soldering, but dealing with a fabrication company sounds like a headache and is a huge barrier of entry for me. You should consider buying them in bulk and selling them =]

pikkmarg, 2024-05-10:
Since all of this is Open Sourced then figuring out the production, investment and operational costs that would entail with mass producing the project could be calculated. 

The beauty of open source is that anyone can build upon the initial work. Product development is a really complicated task and mass producing anything will bring forth many legal and customer support challenges which need a lot of capital.

If you have the capital, time and means to do so then go ahead and produce a ton of them to sell if you feel like there is enough interest to for the volume. Just read the license and go about it in a respectful manner.

As this is an active project then changes might occur and updates made to the design. If you have thousands of units sitting in a warehouse while the project found a more reliable way of doing task X, then moving the stuff will probably be harder.

The other beauty of open source is community and there are group buys with people putting together the devices too. 

For example: https://github.com/rbaron/b-parasite/discussions/173

rbaron, 2024-05-10:
@jordanlambrecht I've been thinking about that more seriously recently. I would love to get a batch going. I haven't done that because selling hardware is hard. Testing, packaging, taxes, dealing with returns. But I think doing informal group orders may be a good first approach for me.

Many OSHW electronics people I look up to go the crowdsupply.com route. I understand they abstract away all these hurdles. That's also something I would consider applying to if there's interest.

jordanlambrecht, 2024-05-14:
@rbaron I think that's a great idea. You've put a lot of effort into this project, and I think it's more than fair to open up a revenue stream with it. Hell, even offering the 3D printed chasis could work.

I honestly do view it as removing a barrier of entry for a lot of people. Not everyone can afford a 3D printer, and a lot of people most likely get stuck at the board manufacturing part. If you remove those two roadblocks, this goes from being an advanced-level project to an easy one with an amazing entry point for those who are just getting started with the hobby, parents/teachers teaching kids about hardware/IoT, botanists that want to implement it, etc. The more people that are exposed to open-source projects like this, the further the ideology and importance of open-source spreads.

dcgrove, 2024-05-15:
I would be in for a handful.

fabian727, 2024-05-29:
@dcgrove  then go for it. as posted by pikkmarg there are always some devices spare:

For example: https://github.com/rbaron/b-parasite/discussions/173

vniehues, 2024-07-16:
I recently ordered 5 pieces from JLCPCB including all the components (front) and assembly (front) for around 25€ per piece including shipping, taxes and customs. 
Turn around time was around 10 days from starting the process to the pieces arriving at my doorstep in Germany. 

I ordered the battery clips from a local company for 0.80€ per piece and the J-Link Edu Mini for 60€ (but I now have a new toy ;) ). 
I also ordered misc stuff like pogo clamps and breakout boards for a total of 10€
Flashing was smooth once I had the hardware setup correctly (connect a 3.3v source to the EDU Mini vRef pin so it outputs 3.3v instead of 1.2v). 

I think I will order more PCBs from JLCPCB in the near future since I have many more plants to tend to. 

Before anyone asks: I don't really feel comfortable fronting the money for a bulk order and selling them. Neither am I willing to collect money beforehand and be responsible if something goes wrong. 
I don't even know @rbaron's stance on anybody selling these

npodbielski, 2024-07-17:
@vniehues thanks for info. 
I asked company that I was planning to order from, how much it would cost to order 100 and 1k units. They got back to me with values of 30 and 20euros - not bad. Probably they need to do some preparation for this PCB and it costs that much. After that price should go down. I ordered 10 for now since I do not need that much and I want to test how this gonna work with Raspbery Pi zero. 

> J-Link Edu Mini for 60€
Wow, that costs even more here. Is this only one compatible programmer for b-parasite?

ryanrolds, 2024-08-02:
I use BlackMagic probes to program. There is a learning curve to setting them up, but it's not that bad.

I've used these devices with RPi Zero, Zero 2, 3, 4, and 5 with 100% success.

ryanrolds, 2024-08-05:
I'm also considering doing a large batch order and making some available for sale. Are there any obligations to the FCC to do this? It does look like the module is certifcation with the FCC, but I'm under the impression that going the SDoC route still requires testing and a report on-file.

ryanrolds, 2024-10-27:
Completing the required testing to sell in the US, a manual w/ regulatory verbage, and setting up a  shipping station took some time. I've put a [a store up](https://shop.plantranger.com/products/bparasite-200) and priced the devices as close to cost as I can and still be willing to do the work required to program and fulfill orders.

I'm currently only able to sell in the US. If someone finds the Canadian IC ID for the module, then I can get talk to the lab and go through the process to get it approved. If the module is not already approved for use in Canada, it will require an additional ~$8,000 go through the full process. I have no experience with getting devices approved in the EU. I will be doing some additional research into that and will go through the process if it's feasable.

When the ~10 devices I have left in this batch sell, I will order a batch of 150 units, which will reduce the cost further. Questions and feedback are welcome.

jordanlambrecht, 2024-11-07:
@ryanrolds If you'd like, I can design some enclosures for it for 3D printing!

jordanlambrecht, 2024-11-07:
Also thank you for doing this. This is awesome.

ryanrolds, 2024-11-10:
You're welcome. I don't have a need for cases right now. But, other may want someone to make cases.

ryanrolds, 2024-12-12:
After selling 4 of the 10 devices I had laying around, I ordered another 100 device. The [price has been dropped](https://shop.plantranger.com/products/bparasite-200) to $11.31/device (was $15.89). Happy Holidays. 

@rbaron, I want to call out that JLCPB appears to be out of the temp+humidity sensor (C194656). I had to buy units that another customer was selling on the site to have enough for the order. Have you seen this component out before?

rbaron, 2024-12-14:
@ryanrolds it happens.

I will briefly raise the concerns I already voiced in [this comment](https://github.com/rbaron/b-parasite/pull/214#issuecomment-2329758133). Please correct me if I am wrong, but I understand your goal is to bundle b-parasite as part of your own smart home product. As you scale up your production to 100 boards as middle step toward that goal, I ask that you please be mindful that a handful of volunteers and I invest our free time in not only developing the project but, more importantly, providing hardware and software support to open source enthusiasts. I ask that you please have your own channel to provide support to your paying customers, now and specially once you sell b-parasite as part of your products. Not only would it not be feasible for us to do so, but I imagine a not lot of us -- myself included -- would think it's fair to provide volunteer support for other business' customers.

ryanrolds, 2024-12-16:
It's helpful to know that component has outages. Thank you.

I 100% agree that the community shouldn't be asked or expected to provide support for the devices in the bundled product. I promise that support channels and clear directions to the appropriate channels will be in-place before the beta starts. Currently there are 5 hoseholds using the alpha and they are all local personal friends of mine; none of them will be reaching out to the community. 

I want to help support the community and I hope that the community devices I am currently selling at are helping people. I stuggle with soldering and have ruined a few devices and chips. If at any point you feel the community is being burdned by my project please tell me. I will do everything in my power to address the issue, including switching to another device. I don't think we are at that point, but it's not off the table. 

I value the work you and the rest of the community have put into this project. I hope you view me as a contributor and a member of the community too. I appreciate that you're expressing clear and direct concerns. Thank you.

rbaron, 2024-12-16:
@ryanrolds thank you taking my concerns seriously and for your reassuring response. I appreciate that. I am perfectly fine with your project under the conditions you described. I chose a permissive license for the project and I am still happy about it. While I really hope to one day be able to provide b-parasites directly to users myself as well, I appreciate community members addressing the demand in the meantime. I wish you nothing but success in your project, and I look forward to your continued contributions.