rbaron/b-parasite #201: Issue with zigbee version

ko-maren, 2024-05-13:
Hi,

We found this project a few month ago and ordered some PCRs, NRF52 and components. In total we have 6 boards. Sadly, it seems like our soldering wasn't that perfect (I don't know what else it could be), because some of the board don't work. Hopefully one of you could help, to narrow down our issue.

Some facts about the boards and software:

- Hardware version of the boards are: 2.0.0
- Blinky program works on all boards
- Zigbee program does not work on three of them. The other three boards are fine and connected to HA
- Zigbee is build with tool kit v2.5.2

We already tried to re-solder the oscillator.

Thanks in advance!

rbaron, 2024-05-13:
Hi @ko-maren,

It can be difficult to diagnose hardware assembly issues, but I'm happy to try. Can you provide some logs from the boards that are not working?

Could you also send a photo of one of the assembled faulty boards?

ko-maren, 2024-05-14:
Hi @rbaron,

Thanks for your fast response!

I'm using NRF52 via VS Code for the first time, and wasn't aware of the option of showing the logs. Seems like all sensors are not working or at least don't communicate with the chip. After this log, I'm thinking if it makes sense to re-solder the chip, but I'm not sure if it is worth the effort or if it could be something else as well?

```
*** Booting nRF Connect SDK v2.5.2 ***
[00:00:00.003,540] <inf> littlefs: LittleFS version 2.5, disk version 2.0
[00:00:00.003,753] <inf> littlefs: FS at flash-controller@4001e000:0xfa000 is 6 0x1000-byte blocks with 512 cycle
[00:00:00.003,753] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.045,471] <err> i2c_nrfx_twi: Error 0x0BAE0002 occurred for message 0
[00:00:00.045,501] <err> shtc3: Error -5:  in /home/maren/Documents/b-parasite/code/nrf-connect/prstlib/src/shtc3.c:19
[00:00:00.045,532] <err> shtc3: Error -5:  in /home/maren/Documents/b-parasite/code/nrf-connect/prstlib/src/shtc3.c:27
[00:00:00.045,532] <err> sensors: Error -5:  in /home/maren/Documents/b-parasite/code/nrf-connect/prstlib/src/sensors.c:15
ASSERTION FAIL [!prst_sensors_read_all(&sensors)] @ ../src/main.c:304
[00:00:00.045,593] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000130  r2/a3:  0x00000005
[00:00:00.045,623] <err> os: r3/a4:  0x20001350 r12/ip:  0x00000014 r14/lr:  0x00007935
[00:00:00.045,623] <err> os:  xpsr:  0x41000000
[00:00:00.045,654] <err> os: s[ 0]:  0x00000001  s[ 1]:  0x200028d8  s[ 2]:  0x00000000  s[ 3]:  0x00044d49
[00:00:00.045,654] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x0004e5d3  s[ 7]:  0x2000d104
[00:00:00.045,715] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00044ad7  s[10]:  0x0004e5d3  s[11]:  0x2000d104
[00:00:00.045,715] <err> os: s[12]:  0x0004e617  s[13]:  0x0000792b  s[14]:  0x0004e5d3  s[15]:  0x0004e938
[00:00:00.045,745] <err> os: fpscr:  0x0004e617
[00:00:00.045,745] <err> os: Faulting instruction address (r15/pc): 0x00044ac4
[00:00:00.045,806] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:00.045,837] <err> os: Current thread: 0x200028d8 (unknown)
[00:00:00.600,494] <err> fatal_error: Resetting system
[00:00:00.000,457] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
*** Booting nRF Connect SDK v2.5.2 ***
```

The third board I currently don't have by hand, but I took a photo of two of faulty boards:

[image]

rbaron, 2024-05-14:
Tks @ko-maren,

From the logs the issue seem to be related to the SHT3 temp/humidity sensor. This one really is tricky to assemble manually, specially if you're using a soldering iron.

You can try to see if there's a short and test the continuity from the sensor pins to the MCU pin. Touching up the solder joins with the iron for a few seconds can help. In the worst case, if that doesn't work, we can try to disable the sensor, so at least the rest of the components and the soil sensing works. Let me know and I can help you with that.

ko-maren, 2024-05-15:
Thanks @rbaron, 

we re-soldered the SHTC3 on all three boards. One of the boards worked afterwards, so it was an issue with this component. On the other two boards we also re-soldered the NRF52. Sadly it helped only one of them back to live.
We are still trying to figure out, what the issue with the third one is, but we currently don't have a debugger by hand, so no change to see the log. I can keep you updated if we can get it back to live.