rbaron/b-parasite #202: Power-Consumption in deep-sleep mode

Exeu, 2024-05-18:
Hey,
i decided to measure the power-consumption of the board using an STLinkPWR.
Here are the results of the stock ble firmware flashed on it:

[image]

The Minimum Power-Consumption i was measuring (during deepsleep) was aroung 40 uA.
It seems this is 10 times higher than what you measured according to the readme of the FW.

Do you have an idea what this could be?

Exeu, 2024-05-18:
[image]

Here a more detailed view of the deepsleep state

rbaron, 2024-05-18:
@Exeu No idea. I redid the measurements last week for the BLE sample and using the PPK 2 I still see around 3.6 uA average. I don't know how the one you're using works, but could it be a resolution thing?

Btw starting a RTT logger/debugger messes up with measurement, make sure you do the measurement without that. 

Let me know what you find, I'm quite curious. Have you tried measuring other boards in the microamp region?

Exeu, 2024-05-18:
> @Exeu No idea. I redid the measurements last week for the BLE sample and using the PPK 2 I still see around 3.6 uA average. I don't know how the one you're using works, but could it be a resolution setting thing?
> 
> Btw starting a RTT logger/debugger messes up with measurement, make sure you do the measurement without that. 
> 
> Let me know what you find, I'm quite curious. Have you tried measuring other boards in the microamp region?

Thanks for that fast answer! Good hint. If I am back home I will go through the fw again and see if I accidentally changed sth. I will let you know

rbaron, 2024-05-18:
Awesome, tks! If you have access to another b-parasite board, it would be cool to run the profile there too and see if you can reproduce it, in case you do manage to measure single digit micro amps with a different board or just across a 1M resistor. I would be very concerned if we see that high of a deep sleep current consistently 😰

Exeu, 2024-05-19:
Soo i first of all i now checked FW config. It is as follows:

```
CONFIG_LOG=n
CONFIG_PWM=y
CONFIG_CBPRINTF_FP_SUPPORT=y
CONFIG_I2C=y
CONFIG_ADC=y
CONFIG_GPIO=y

CONFIG_BT=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_DEVICE_NAME="mfh"
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y

CONFIG_SERIAL=n

CONFIG_USE_SEGGER_RTT=n

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y

CONFIG_ASSERT=y

# Application config - see all options in Kconfig.
# CONFIG_PRST_BLE_ENCODING_BTHOME_V2=y
# CONFIG_PRST_SLEEP_DURATION_SEC=1

# prstlib config - ser all options in prstlib/Kconfig.
CONFIG_PRSTLIB_LOG_LEVEL_DBG=n
``` 
So it seems no RTT and Debug enabled.

@rbaron do you see anything suspicious? I will now have a look on another device

Exeu, 2024-05-19:
Uhh i used another board now and the results are much better!

Overall including transmit:
[image]

Deepsleep:
[image]

rbaron, 2024-05-19:
@Exeu I'm happy to hear you managed to measure around the expected values. Just to double check, when you say another board it's another b-parasite board right (and not some different hardware altogether)?

Exeu, 2024-05-19:
@rbaron it's frome the same assembly charge. One of 5 seems to be malfunctioning. Ordered 5 boards from jlc including assembly so they should be identical