rbaron/b-parasite #204: Coating instructions

Pferdebockwurst, 2024-06-01:
I bought a can of Plastik 70 Super but I'm really not sure where to apply the coating. @rbaron unfortunately, your Twitter thread about the coating process isn't accessible anymore. Can someone please tell me which parts of the sensor must be excluded from the coating?

rbaron, 2024-06-01:
Sadly I've ditched twitter for the most part. But this is the video I had there:

https://github.com/rbaron/b-parasite/assets/1573409/410142f7-b048-4441-b6cc-818e977ba255

And this is the result I aim for:

[image]

For the record I know little about conformal coating. I am sure there are less megalomaniacal ways to achieve better results. But I've had good experience with this, and more importantly, fun building the spinning jig (#63).

I started #79 to gather more ideas and knowledge on the topic.

Pferdebockwurst, 2024-06-01:
> Sadly I've ditched twitter for the most part. But this is the video I had there:
> coating.mov
> 
> And this is the result I aim for:
> 
> ![IMG_9284](https://private-user-images.githubusercontent.com/1573409/335815210-430d5f73-ceed-4c1c-a3bb-f8bfe38ae66a.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyNjEyMTEsIm5iZiI6MTcxNzI2MDkxMSwicGF0aCI6Ii8xNTczNDA5LzMzNTgxNTIxMC00MzBkNWY3My1jZWVkLTRjMWMtYTNiYi1mOGJmZTM4YWU2NmEuSlBHP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDFUMTY1NTExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODM2YTJlMjJhZmQxNWM3NGQ0OWM0MDI0MmU3NDljNGEzOTNlYTUyYzZhOGUwMzczNDYxMmI0MTk1NmIyYjc2YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.giaU5WjM4v3LOX2Ia8ZB4nYejpUxyuMVHe_uRWARs5Q)
> 
> For the record I know little about conformal coating. I am sure there are less megalomaniacal ways to achieve better results. But I've had good experience with this, and more importantly, fun building the spinning jig (#63).
> 
> I started #79 to gather more ideas and knowledge on the topic.

Thanks. Great idea! So you didn't mask anything on the PCB? Looks like you even sprayed on the chips and battery holder?!

rbaron, 2024-06-01:
@Pferdebockwurst lately I've been masking by holding a piece of cardboard so I only spray the bottom half. I've also gone wild in the past and sprayed everything, then removed the coating from the batt clip with a cotton swab and rubbing alcohol. While everything works, I haven't measured the impact on the temp sensor (it has an opening that likely shouldn't be filled...). But these days I just coat the bottom half.

otsoni, 2024-10-05:
@rbaron, is that turning machine 3D-printed and if so, is there files available somewhere?

otsoni, 2024-10-05:
> @rbaron, is that turning machine 3D-printed and if so, is there files available somewhere?

Sorry, I just found it!