rbaron/b-parasite #206: Battery and soil sensors invalid readings

seime, 2024-06-02:
Hi,

thank you for your great project!

I ordered a handful of these `v2.0.0` boards pre-soldered from JLCPCB (all components except the battery clip), and I've flashed them with the BLE example. However, the readings from the battery and soil sensor is always 0mV.

Here is an excerpt from the log:

```
00:03:13.918,640] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 960.00 | Raw 4294967294 | Batt: 0.00 | Dry: 334.00 | Wet: 299.00
[00:03:13.918,701] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 960.00 | Raw 4294967294 | Batt: 0.00 | Dry: 334.00 | Wet: 299.00
[00:03:13.928,863] <dbg> adc: prst_adc_photo_read: Read phototransistor: 142 lx | 0.02 V
[00:03:13.951,904] <dbg> shtc3: prst_shtc3_read: Read temp: 28.897552 oC (28)
[00:03:13.951,934] <dbg> shtc3: prst_shtc3_read: Read humi: 39 %
[00:03:13.951,965] <dbg> sensors: prst_sensors_read_all: Batt: 0 mV (0.00%)
[00:03:13.951,965] <dbg> sensors: prst_sensors_read_all: Soil: 100 % (0 mV)
[00:03:13.951,995] <dbg> sensors: prst_sensors_read_all: Photo: 142 lx (0 mV)
[00:03:13.951,995] <dbg> sensors: prst_sensors_read_all: Temp: 28.897552 oC
[00:03:13.952,026] <dbg> sensors: prst_sensors_read_all: Humi: 39 %
[00:03:13.952,056] <dbg> sensors: prst_sensors_read_all: --------------------------------------------------
[00:03:16.983,489] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 965.71 | Raw 4294967292 | Batt: 0.00 | Dry: 334.00 | Wet: 299.00
[00:03:16.983,551] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 965.71 | Raw 4294967292 | Batt: 0.00 | Dry: 334.00 | Wet: 299.00
[00:03:16.993,713] <dbg> adc: prst_adc_photo_read: Read phototransistor: 0 lx | 0.00 V
[00:03:17.016,754] <dbg> shtc3: prst_shtc3_read: Read temp: 28.878860 oC (28)
[00:03:17.016,784] <dbg> shtc3: prst_shtc3_read: Read humi: 39 %
[00:03:17.016,815] <dbg> sensors: prst_sensors_read_all: Batt: 0 mV (0.00%)
[00:03:17.016,815] <dbg> sensors: prst_sensors_read_all: Soil: 100 % (0 mV)
[00:03:17.016,845] <dbg> sensors: prst_sensors_read_all: Photo: 0 lx (0 mV)
[00:03:17.016,845] <dbg> sensors: prst_sensors_read_all: Temp: 28.878860 oC
[00:03:17.016,876] <dbg> sensors: prst_sensors_read_all: Humi: 39 %
[00:03:17.016,906] <dbg> sensors: prst_sensors_read_all: --------------------------------------------------
```

Debug steps taken so far:

- Tried 3 different boards, all with identical results
- Tried with and without the CR2032 battery installed (3v from JLink disconnected)
- Ensured there is connectivity form pin 3 (SENS_OUT) to R6

Reading of battery seen from the debugger:

[image]

Any suggestions on how to proceed debugging?

Thanks again for this cool project!

drspangle, 2024-06-02:
I would triple-check the soldering from the ESP to the main board. I had a similar situation with a few of mine and after resoldering (one of them took a few tries) it was successful.

seime, 2024-06-02:
Thanks, I did a triple check on one of the boards. All pads _apparently_ connected (by measuring). Tried resoldering one side, but as I dont have suitable equipment for SMD I'm afraid I'll do more damage than good.

The reason I ordered pre-assembled on JLCPCB was to ensure proper soldering. My expectations were that all pads are actually soldered, but as I do not have any prior experience ordering custom PCBs I might have expected too much?

rbaron, 2024-06-02:
Hi @seime, tks for the kind words. Glad to hear you managed to order and flash the boards.

From the logs, it looks like all ADC functionality is acting weird -- battery, soil moisture and illuminance are all reading 0 V. The battery measurement is internal to the nRF52 chip, so I would first rule out other issues than soldering -- although this issue has happened to me as well. If the module is powered up and running (which it is), I would expect it to read a valid batt voltage.

We support both `nRF52840` and the `nRF52833` modules. I would check that you built the sample for the correct variant and version (e.g.: `bparasite_nrf52840` & 2.0.0). This would be my first guess. I would also double check the module that JLCPCB soldered on the board (e.g.: `E73-2G4M08S1C` `E73-2G4M08S1E`). Just to be sure: if you're providing external power, do not use a battery and vice-versa.

Let us know how it goes, and I can try thinking of other possible issues.

seime, 2024-06-02:
Thanks for your prompt reply @rbaron .

I'm using the nRF52840 in the E73-2G4M08S1C module.

I tried to change the overlay file to the bparasite specific one, and then at least battery voltage came alive:

This is the config I tried:

[image]

```west build --build-dir /opt/openhab/b-parasite/code/nrf-connect/samples/ble/build /opt/openhab/b-parasite/code/nrf-connect/samples/ble --pristine --board bparasite_nrf52840@2.0.0 --no-sysbuild -- -DNCS_TOOLCHAIN_VERSION=NONE -DBOARD_ROOT=/opt/openhab/b-parasite/code/nrf-connect/prstlib -DCACHED_CONF_FILE=/opt/openhab/b-parasite/code/nrf-connect/samples/ble/prj.conf -DDTC_OVERLAY_FILE=/opt/openhab/b-parasite/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52840/bparasite_nrf52840_2_0_0.overlay```

Output is now

```
[00:08:20.435,577] <dbg> sensors: prst_sensors_read_all: --------------------------------------------------
[00:08:23.467,010] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: -4.06 | Raw 546 | Batt: 3.30 | Dry: 530.42 | Wet: 146.70
[00:08:23.467,071] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: -4.06 | Raw 546 | Batt: 3.30 | Dry: 530.42 | Wet: 146.70
[00:08:23.477,233] <dbg> adc: prst_adc_photo_read: Read phototransistor: 8498 lx | 1.43 V
[00:08:23.500,244] <dbg> shtc3: prst_shtc3_read: Read temp: 29.129868 oC (29)
[00:08:23.500,274] <dbg> shtc3: prst_shtc3_read: Read humi: 43 %
[00:08:23.500,305] <dbg> sensors: prst_sensors_read_all: Batt: 3304 mV (100.00%)
[00:08:23.500,305] <dbg> sensors: prst_sensors_read_all: Soil: 0 % (0 mV)
[00:08:23.500,335] <dbg> sensors: prst_sensors_read_all: Photo: 8498 lx (0 mV)
[00:08:23.500,335] <dbg> sensors: prst_sensors_read_all: Temp: 29.129868 oC
```

Ie the battery voltage is now also working. The light sensor was working before as well - that is the lux value is updating but the voltage is always reported at 0... Soil sensor still reading 0mV.

Newbie question to nrf programming: The device overlays, is it supposed to be just 1 in our case - or more?

rbaron, 2024-06-02:
It looks like it's a good step in the right direction. I have a feeling it could be actually working now, but something is off in the logging of float values (compare the 100% soil moisture before to 0% now; and also the 1.43 V for the illuminance vs. 0.02 V before). You can quickly check by resetting the board while squeezing the sensor in your hand -- the soil moisture should say 50%+.

The float print issue could be related to `CONFIG_NEWLIB_LIBC` and [`CONFIG_NEWLIB_LIBC_FLOAT_PRINTF`](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/kconfig/index.html#!CONFIG_NEWLIB_LIBC_FLOAT_PRINTF), but since [these should be enabled in the ble sample](https://github.com/rbaron/b-parasite/blob/main/code/nrf-connect/samples/ble/prj.conf#L18), I think there's something else going on with the build that is preventing these configs from working.

In VSCode, I've never had to manually specify the overlay file -- it should be automatically loaded by west from the `--board` switch. I.e.: `--board bparasite_nrf52840@2.0.0` should auto load `prstlib/boards/arm/bparasite_nrf52840/bparasite_nrf52840_2_0_0.overlay`. Are you opening VSCode from the `ble.code-workspace` file?

seime, 2024-06-03:
Good morning,

Apparently I have done something wrong when I opened the project and built the firmware. After reopening the workspace file it works "better".

This is what I get when I press a finger on the board;

`          ``
[00:00:13.128,875] <dbg> sensors: prst_sensors_read_all: --------------------------------------------------
[00:00:16.160,278] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 42.59 | Raw 367 | Batt: 3.30 | Dry: 530.39 | Wet: 146.73
[00:00:16.160,339] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 42.59 | Raw 367 | Batt: 3.30 | Dry: 530.39 | Wet: 146.73
[00:00:16.160,400] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 42.59 | Raw 367 | Batt: 3.30 | Dry: 530.39 | Wet: 146.73
[00:00:16.160,430] <dbg> adc: get_soil_moisture_percent: Read soil moisture 2: 42.59 | Raw 367 | Batt: 3.30 | Dry: 530.39 | Wet: 146.73
[00:00:16.170,593] <dbg> adc: prst_adc_photo_read: Read phototransistor: 82 lx | 0.01 V
[00:00:16.193,634] <dbg> shtc3: prst_shtc3_read: Read temp: 26.817398 oC (26)
[00:00:16.193,664] <dbg> shtc3: prst_shtc3_read: Read humi: 37 %
[00:00:16.193,695] <dbg> sensors: prst_sensors_read_all: Batt: 3301 mV (100.00%)
[00:00:16.193,695] <dbg> sensors: prst_sensors_read_all: Soil: 43 % (0 mV)
[00:00:16.193,725] <dbg> sensors: prst_sensors_read_all: Photo: 82 lx (0 mV)
[00:00:16.193,756] <dbg> sensors: prst_sensors_read_all: Temp: 26.817398 oC
[00:00:16.193,756] <dbg> sensors: prst_sensors_read_all: Humi: 37 %
[00:00:16.193,786] <dbg> sensors: prst_sensors_read_all: --------------------------------------------------
```

The debug log always logs `0 mV` on the `Soil` and `Photo` sensor, even if the percentage and lx values change

Edit: Photo logging is using soil adc reading; fixed here: https://github.com/rbaron/b-parasite/pull/207

Still the soil reading seems to not get converted to millivolts. I can see the raw value changed based on finger pressure:

![CleanShot 2024-06-03 at 10 14 16](https://github.com/rbaron/b-parasite/assets/1565156/007efdf2-993c-4ae9-bcec-2d3084977129)

Still something wrong with the overlay?

rbaron, 2024-06-04:
Hey @seime, thanks for the PR! Nice catch.

For the `0 mV` in the soil moisture reading, the situation is a bit more involved. We sample the ADC peripheral at the soil moisture sensor pin and get back a "raw" value. In our case, this is a 10-bit value from 0-1024.

Then we call [adc_raw_to_millivolts_dt](https://docs.zephyrproject.org/latest/hardware/peripherals/adc.html#c.adc_raw_to_millivolts_dt) that converts the raw value into a millivolt value, taking into account how the ADC channel is configured.

For the battery and the phototransistor, this works fine, because they use the 0.6 internal reference in their channels ([example](https://github.com/rbaron/b-parasite/blob/ccdbaaf142c0d5e12d5ab070e809df4f5cd843d6/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52840/bparasite_nrf52840.dts#L133)). For the soil moisture sampling, I configured it to use the `VDD` as a reference, to try and make the readings more or less agnostic to the current battery level ([here](https://github.com/rbaron/b-parasite/blob/ccdbaaf142c0d5e12d5ab070e809df4f5cd843d6/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52840/bparasite_nrf52840.dts#L121)). 

But it looks like then `adc_raw_to_millivolts_dt` doesn't work well and returns 0 every time. I did a quick test by changing the ref to `ADC_REF_INTERNAL` and it does work. Maybe there's a workaround, but either way it doesn't impact us that much, other than that logging line -- the soil moisture estimation is done over the raw values anyway.

So for now I will suggest that we remove the `millivolts` from the line that logs soil moisture in `sensors.c`. Let me know if you want to do that in your PR as well, otherwise I will send another one.

Tks!

seime, 2024-06-04:
Thanks @rbaron, I've updated the PR removing the soil voltage reading.

Thanks again for your awesome support in resolving my issue!

closed 2024-06-04 by seime

rbaron, 2024-06-04:
Tks!