rbaron/b-parasite #213: Zigbee Rejoin Network

HartD92, 2024-08-01:
I have a number of devices running the latest (non-OTA) zigbee build. While they have been reasonably stable, whenever a device has an issue or drops off the network, the only way to get it to rejoin is to reflash the device. Rebooting or shorting the reset pin does not seem to get it back, even though it flashes the rejoin LED pattern.

hw version is 1.2

HartD92, 2024-08-01:
I tried a new battery and it is able to report again. The LED was still quite bright on the old battery. Why would the battery be enough to power the device and run the LED, but not be able to report back?

rbaron, 2024-08-01:
@HartD92 I can't say I've experienced this issue so far running on ZHA. It would be interesting to capture some logs from your boards when it happens.

Out of curiosity, on what platform/hardware are you running your coordinator?

HartD92, 2024-08-01:
Hi, it's running on Zigbee2MQTT, with a Conbee II radio. I'm using a black magic probe to flash the boards, and I've not been able yet to figure out how to actually get the debug information from them.

oleo65, 2024-08-26:
Your second comment https://github.com/rbaron/b-parasite/issues/213#issuecomment-2263455875 gives the hint to at least explain it for me.

During transmit the MCU uses a lot more power than during idle operation. Pairing and rejoining the network is especially demanding. This is what I observed during power measurements of the devices. If your battery is already low, it simply might not provide enough current to the device during rejoining and the device will fail. This might result in a power cycle or some unpredictable state.

You should however be able to recover from this without reflashing the device by simply inserting a fresh coin cell. I did not have any of my devices needing a reflash due to power issues. 🤔