rbaron/b-parasite #215: WIKI link pogo pins not avilable

eiten, 2024-09-10:
In the wiki on page https://github.com/rbaron/b-parasite/wiki/How-to-Flash-the-Firmware-Samples#soldering-vs-using-pogo-pins there is a link to X for the pogo pins. But the X Post https://x.com/rbaron_/status/1382226349088194560 is not available publicly...

rbaron, 2024-09-10:
Nice catch, thanks @eiten. I updated the wiki with the direct embedded video. Let me know if that works.

closed 2024-09-10 by rbaron

eiten, 2024-09-11:
Well, no... in Firefox (windows 10, ubuntu 22.04) it shows that it did not find a suitable video with supported format and MIME type 
[image]
Edge (windows 11) and chrome (ubuntu 22.04) show the video window, but playing does not work (grayed out controls)
[image]

I think the problem is the QuickTime format of the video, I suppose only supported by Apple devices. I converted the video to webm and mp4 with settings for web, they should work:

https://github.com/user-attachments/assets/24300131-fa35-453f-aa76-d40abd2491b5

https://github.com/user-attachments/assets/9eb25976-7ec4-4ed1-84cf-cc9a35660156

And maybe you add a link to the ali article (maybe an affilate-Link ;)). I don't need it because i got Pogo-clamps in every count and pitch but others may be lucky if they don't have to search for them.

Regards, Edi

eiten, 2024-09-14:
> I'm always frustrated when they eventually go stale.

Well, that's true...