rbaron/b-parasite #221: ZBOSS SDK upgrade from 2.5 to 2.7 fails to migrate zigbee network info

elratt0r, 2024-11-14:
When upgrading from SDK 2.5 to 2.7 the chip keeps rebooting with the message `<err> zboss_osif: ZBOSS fatal error occured`.
After some debugging it turned out to be the `zb_nvram_write_data` called by `migrate_curr_page_to_new_version()`.
(See also: https://devzone.nordicsemi.com/f/nordic-q-a/113710/zboss_osif-zboss-fatal-error-occurred-when-migrating-sdk-from-2-3-0-to-2-7-0)

No solution to migrate atm. Depending on chosen reset method it is possible to factory-reset the chip (SW1 method does not work because reboot occurs before 5sec timeout). In that case you have to use `nrfjprog --eraseall` and repair the chip.

Suggestion/Topic to discuss:
Make SW1 factory reset a blocking event, so it can kick before the ZBOSS bootup.

rbaron, 2024-11-14:
Thanks for reporting this @elratt0r. 

This should impact people who previously flashed their devices using the older `2.5` SDK, paired them with a coordinator and _then_ reflashed them with a `2.7` build. Ideally the pairing information should be kept (some welcome news to me), but there is a ZBOSS issue with the automatic update of the stored pairing info. Is this correct?

So if other users are reading this, it should not impact anyone who's flashing their device for the first time and/or do a wipe + flash. I'm also dropping it here that you can fully erase the flash while flashing a new firmware directly via the VSCode extension (it calls `nrfjprog --eraseall` under the hood):

[image]

elratt0r, 2024-11-14:
> This should impact people who previously flashed their devices using the older `2.5` SDK, paired them with a coordinator and _then_ reflashed them with a `2.7` build. Ideally the pairing information [...]

That is correct.

I will keep an eye on this topic on Nordic side, as it also affects the OTA PR.