rbaron/b-parasite #226: Solar node?

samuk, 2025-03-10:
I'm looking into solar options, is the b-parasite 3.3v tolerant?

Thinking I could a ML2032  and https://www.tindie.com/products/bobricius/usb-lithium-coin-cell-battery-charger-ml2032/ connected to a small 5v panel.

The parasite would get 3.3v during the day when charging

happytm, 2025-05-01:
Isn't your design at https://github.com/rbaron/lir2450-charger cheaper option?

rbaron, 2025-05-02:
@happytm It could work, but not by itself. You'd still need a [4.2, 3.7] -> 3v3 regulator, batt overdischarge protection etc. But samuk's approach would also need these afaict, as both LIR2450 and LIR2032 go up to 4.2 V. I haven't tried either with high impedance solar cells as input, but I think it could work. Just make sure to select a regulator that has very little quiescent current (ideal single digit microamp).

Another interesting approach is to use an energy harvesting solution like the [`BQ25570`](https://www.ti.com/product/BQ25570) or [AEM10941](https://eu.mouser.com/datasheet/2/1087/e_peas_DS_AEM10941_QFN28_v2_1-3512266.pdf). These harvest power from tiny, milliwatt solar cells starting from as low as 0.3 V. They ICs are relatively expensive, and you'd still need to design a bunch of auxiliary circuitry, but for an application that uses on average microwatts like b-parasite, I think they are a great fit.

samuk, 2025-05-02:
Just spotted these that look interesting 

https://medium.com/@emmaodunlade/exploring-smks-self-contained-power-supply-coin-battery-module-a-game-changer-in-energy-storage-b4c87875b49b

I suspect they won't be cheap

happytm, 2025-05-03:
These little batteries are interesting if they can be used for low power micros.

https://www.youtube.com/watch?v=lQBfx6_fVXM

Thanks.

samuk, 2025-05-09:
So one option might be to edit https://github.com/rbaron/lir2450-charger

To add a https://www.ti.com/product/TPS63051#order-quality and a pair of +/-female headers.

Then make a minor edit the bparasite to have a couple of +/-male pins. 

You could then connect a 5v usb panel to the lir charger?

If people want replaceable coin-cell they get a 'standard b-parasite' with the two pins they don't use. If they want solar they buy the extra board and plug it in?

Does that seem halfway sensible?

samuk, 2025-05-11:
Is 3.8v tolerable? Or pushing it? https://voltaicsystems.com/Solar-System-Lithium-Ion-Capacitor

samuk, 2025-05-19:
[image]

I had a little play around with[ this pcb](https://easyeda.com/editor#id=31e6f3bcd26b4b6f81ee6815250d6a68) which is just a battery holder and 3v regulator. It would take a [USB chargeable 16340](https://vi.aliexpress.com/item/1005008601347462.html)

I'm thinking if I get some b-parasites made up with a female header then it could just press fit into the header? Optionally connect a 5v USB panel to the battery.

I am assuming that turning the pads into TH holes for a header will be unproblematic?

[image]

rbaron, 2025-05-20:
@samuk the pad -> header should be okay, just make sure you have a plan to connect your programmer to it.

I haven't looked at the charging circuit, but in general make sure the battery has over and undervoltage protection. It's also going to be frequently close to water, so chances of splashing and shorting are high. LiPos can pack a much bigger punch than the CR2032 if things go wrong.