rbaron/b-parasite #231: Battery on same side as components?

samuk, 2025-05-11:
I just had a look at getting some fabricated and the double sided assembly almost doubles the cost.

If I edit the PCB to make part of it wider add a 'battery blob' to the side of it's current position to put the battery on the same side as the components are there things related to the capacitance I need to consider? Or is it a relatively straightforward job.

rbaron, 2025-05-11:
For batches of up to a couple dozen boards, I usually get the top side only assembled by JLC and hand solder the battery clip on the back.

samuk, 2025-05-11:
I appreciate that, but I don't like soldering.

rbaron, 2025-05-11:
Then I think your options are:
- Get double sided PCBA
- Get someone to solder it for you
- Redesign the board to place the batt clip in front. From experience you'll want to keep all the electronics as far from the ground as possible to avoid shorts when watering the plants. It's possible that the current capacitance calibration is good enough, but I can't say for sure

samuk, 2025-05-11:
I guess the other approach is to populate the header pins with male or female headers and provide 3.3v externally. I might have a go at that

closed 2025-05-11 by samuk